DCC-EX/CommandStation-EX #383: Support for Nextion touch HMI displays

VR-Architect, 2024-01-03:
I would like to add support for Nextion touch HMI displays. I am using them for route panels and turnout control. It great simplifies my wiring.

What is the best way for me to add this to DCC-EX? I am a programmer and happy to code it if someone can show me the ropes of the DCC-EX code architecture.

Thanks

habazut, 2024-01-03:
Currently DCCEX has 3 interfaces:
1. I2C / EX-Expander. The nextion could act as "buttons" or "analog input"

2. character stream with DCCEX commands. Serial or TCP/IP. Currently there is only one type of sensor defined: <q 17> <Q 17> for sensor 17 off and on. This can be extended for good reasons, for example to support inputs that are not binary.

3. Withrottle. But withrottle does ONLY support throttle commands.

Then someone did build an MQTT bridge, but I have lost that project a bit out of sight.

Do you have a controlling software like JMRI running on a computer?
How do/can the Nextions communicate?

Harald.

VR-Architect, 2024-01-03:
> Do you have a controlling software like JMRI running on a computer? How do/can the Nextions communicate?

Hi Harold, thanks for the valuable information. I currently have it working with a Mega (without DCC-EX) connected via Serial1 using RX/TX pins. The display sends strings onto the Serial when a button is clicked to toggle Kato turnouts and then when the Arduino throws the turnout, the Arduino sends back a string telling the display to indicate the direction of the turnout. 

In addition, I have buttons for routes. When clicked, the display sends a string indicating which route to toggle. The Arduino, throws all the turnouts in the route, then sends back a string to the display to change the button color indicating the route was activated. 

I can have up to 4 displays connected to the Mega. 

If DCC-EX can't do leverage the Serial communication, then I might be able to put an Attiny85 between DCC-EX and the display to convert I2C <> serial? 

Thanks again.

habazut, 2024-01-04:
Look at https://dcc-ex.com/throttles/tech-reference.html#gsc.tab=0
I think you can do already a lot by triggering routes and operating turnouts (real turnouts and virtual turnout objects) It might be you only need to change what strings to send.
Yes we can parse so many serial lines which we have extra on the microcontroller in question. If more are needed, one might build a serial line concentrator or use TCP/IP.
Harald.