DCC-EX/CommandStation-EX #384: Dcc-ex and Traincontroller

Topolroot, 2024-01-04:
Before asking my obvious question, I want to thank you from the bottom of my heart for your work! You and your whole team are super people! You are doing a good and necessary thing! Thank you! Now the question is, can the DCC-EX command station work with the Traincontroller program? If it can, which controller in the program should I choose? For example, in the Rocrail program, this is a DCCPP controller. Thank you in advance. My mail topolroot@gmail.com

habazut, 2024-01-04:
That is more a question for the guy selling TrainController.

I know that iTrain and JMRI do support DCCEX. And there is RocRail which says they do support DCC++ which is probably enough to run DCCEX as well.

I have heard nice things about the service of iTrain.

I don't use TrainController or iTrain or RocRail. I do use JMRI for decoder programming, not for automation.
You can use the builtin EXRAIL for automation. You may not need a bit software. Depends on your layout
and needs.

Harald.

Topolroot, 2024-01-06:
Thank you very much for such a full detailed answer. I myself actively use
the Rocrail program. Everything works successfully in it. I will try to
contact the developers of the tre controller with a similar question. Thank
you very much again, you are a great fellow!