DCC-EX/CommandStation-EX #400: Installation Error

jaypandith, 2024-03-12:
2024-03-12 14:03:52.972 - ex_installer.git_client - ERROR - Local file changes in pygit2.Repository('/Users/jaypanditharatne/ex-installer/CommandStation-EX/.git/')
2024-03-12 14:03:52.972 - ex_installer.git_client - ERROR - ['data/Tlc5940_r014_2.zip (Added)', 'myAutomation.example.h (Modified)', 'version.h (Modified)']
2024-03-12 14:03:52.976 - ex_installer.select_version_config - ERROR - Local repository file changes: ['data/Tlc5940_r014_2.zip (Added)', 'myAutomation.example.h (Modified)', 'version.h (Modified)']
2024-03-12 14:05:09.252 - ex_installer.ex_installer - CRITICAL - Uncaught exception: Traceback (most recent call last):
  File "tkinter/__init__.py", line 1921, in __call__
  File "customtkinter/windows/widgets/ctk_radiobutton.py", line 387, in invoke
    self._command()
  File "ex_installer/select_version_config.py", line 281, in set_next_config
TypeError: 'NoneType' object is not subscriptable

peteGSX, 2024-03-12:
That error indicates you have made changes to the files myAutomation.example.h and version.h, and you have also added a zip file data/Tlc5940_r014_2.zip.

All files within the EX-Installer directory need to be left alone for EX-Installer to manage, otherwise it will fail, which is clearly outlined in our documentation: https://dcc-ex.com/ex-installer/installing.html

If you haven't edited those files intentionally, the simplest resolution is to delete the directory "/Users/jaypanditharatne/ex-installer/CommandStation-EX/" and try again.

Cheers, Pete

jaypandith, 2024-03-12:
Hi Pete,

Thanks for your prompt response, as instructed  I re-installed DCC-EX, 

Cheers

Jay

peteGSX, 2024-03-12:
Good stuff Jay, does that mean this issue is now resolved?

Cheers, Pete

jaypandith, 2024-03-12:
Morning Pete,Thanks, yes the issue fully resolved and extremely happy with my first attempt with DCC-EX (migrated from Digitrax DCS52).Once again thanks for your feedback,Cheers Jay (from Melbourne AU)Sent from Jay Panditharatne's iPad Pro

peteGSX, 2024-03-12:
Glad to hear it Jay! (from sunny Brissie).

closed 2024-03-12 by peteGSX