DCC-EX/CommandStation-EX #415: Bitdefender detected malicious code in EX-Installer-Win64.exe

movinjim, 2024-08-20:
Windows 10, Bitdefender Total Security, latest release of EXInstaller.

BitDefender gives this error and quarantines EXInstaller, not letting it run. 

The app C:\Users\Jim Lee\Documents\Maker Projects\DCCPlusPlus\EX-Installer-Win64.exe infected with Gen:Suspicious.Cloud.14.@x5@am0Do3p was moved to quarantine. It is recommended that you run a System Scan to make sure your system is clean.

I will not install EXInstaller at this time. Please advise.
Thanks,
Jim Lee

FrightRisk, 2024-08-21:
Windows requires that programs like EX-Installer go through their onerous BitDefender approval process for every single version of software we release. They are not always prompt. That version should be approved soon, but there is nothing wrong with the file. You can prove this yourself by manually submitting the installer or the file to BitDefender on their website. It is a shame they have a warning insinuating a file is "infected" instead of the truth, which is "our cloud based quick check doesn't know what this is, so have labelled is suspicious". You can choose to use it, use a previous version that already has been flagged good by them, or use the Arduino IDE.

movinjim, 2024-08-21:
Thanks for your prompt response and a clear explanation. You are right that it's a shame they report it as infected when it it not.
As I'm wanting to update to the lastest version, I'll likely just use the Arduino IDE. A bit of a pain, but doable.
Jim

pmantoine, 2024-08-22:
Microsoft Windows Defender does not see it as a virus... so you're effectively paying for independent anti-virus company BitDefender to mislead you and result in you being disadvantaged.

Note too that you'll also need to use GitHub Desktop to be able to have access to development versions of DCC-EX which are trivially easy to access with EX-Installer.

habazut, 2024-08-22:
As you are a Bitdefender customer you should have a way to tell them that you do not think that their product made  the right decision. We as the DCCEX community can not contact all 3rd party companies that might decide to label our software in some way. We do that towards the OS vendors.