DCC-EX/CommandStation-EX #434: ESp32s3 compatible installer

Denso1980, 2025-01-16:
I bought a board ESP32duino with ESP32-S3 clone chip https://www.tztstore.com/goods/show-6284.html. I tried to compile the latest version of CS firmware however I have a fatal error end of the esp32-s3 not compatible and unable to flash this board. I tried different settings in Arduino ide 2.3.4 choosing different boards but I was not luck after. Can you modify the installer to be compatible with esp32-s3 too, please?

Ash-4, 2025-01-16:
There are no plans to support other ESP32 variants.  ESP32-WROOM is the one supported for EX-CommandStation.
Join the discussion on the discord server for further questions.
https://dcc-ex.com/support/index.html#preferred-option-discord

Ash-4, 2025-01-16:
https://dcc-ex.com/reference/hardware/microcontrollers/esp32.html#esp32-wroom-32-boards-tested

habazut, 2025-01-16:
Actually the ex-installer is compatible with the ESP32-S3. However the EX-CS product is not and we do not have any plans to make that happen soon. Other products like the DCCInspector might work on the ESP32-S3.

Regards.

closed 2025-03-08 by peteGSX