DCC-EX/CommandStation-EX #442: Missed DCC-EX native API command for Turntable deletion

KonstantinKolesnik, 2025-02-23:
It's possible to add a turntable but no way to delete it.
It seems the Turntable and inherited classes don't have the corresponding method as well.

Please add the missed API command.

Asbelos, 2025-02-25:
Turntables created by <I commands only last until bootup.
Turntables defined in myAutomation are created at bootup, not by executing a sequence.

Thus there is no use cser for deleting a turntable at run time.

KonstantinKolesnik, 2025-02-25:
So it's not possible to store turnouts in EEPROM?

Asbelos, 2025-02-25:
It is, but that's only to be backward compatible with the old dcc++ and only on the cpus that it used to run on ( mega and uno() and dcc++ did not have turntables.
It is strongly discouraged to store anything  in eeprom. 
Keep them in jmri or in myAutomation.h where yiu can see them, edit them, backup and restore etc.

KonstantinKolesnik, 2025-02-26:
So the approach to create turnouts, signals and outputs from my custom throttle software and store them in EEPROM is not recommended?
And in general, a throttle is recommended to work with objects in read-only mode? No create/delete?

Asbelos, 2025-02-26:
The confusing and messy (totally agree) commands are mostly there to satisfy jmri users who want to keep doing it in jmri.
However, there are limitations due to things like ram consumption (uno and Mega in particular) 
 which prevent us from including things like descriptions and we can't dynamically load hardware drivers at run time anyway. 

It's generally much easier to edit the hardware description of your layout on a PC and upload it to the cs. This gives you access to all features and multiple hw types. 
Experimentation with things like turnout servo settings is  best done with EXToolbox app.

closed 2025-03-02 by KonstantinKolesnik