DCC-EX/CommandStation-EX #444: [Feature Request]: Add broadcast event for accessory control

reinder, 2025-03-28:
As a test I connected twice to a CommandStation-EX, one serial and one (wired) network using [Traintastic](https://github.com/traintastic/traintastic). (My open source project for model railway control.)

When you control an engine, the other client(s) are notified, see log below:
```
2025-03-28 09:48:48.853718 dccex_network D2001: TX: <t 1 1 0 1>
2025-03-28 09:48:48.863162 dccex_serial  D2002: RX: <l 1 0 128 0>
2025-03-28 09:48:48.873023 dccex_network D2002: RX: <l 1 0 128 0>
2025-03-28 09:48:48.882254 dccex_network D2002: RX: <T 1 0 1>
2025-03-28 09:48:53.338391 dccex_network D2001: TX: <F 1 0 1>
2025-03-28 09:48:53.347916 dccex_serial  D2002: RX: <l 1 0 128 1>
2025-03-28 09:48:53.354009 dccex_network D2002: RX: <l 1 0 128 1>
2025-03-28 09:48:57.118718 dccex_serial  D2001: TX: <F 1 0 0>
2025-03-28 09:48:57.124332 dccex_serial  D2002: RX: <l 1 0 128 0>
2025-03-28 09:48:57.218955 dccex_network D2002: RX: <l 1 0 128 0>
2025-03-28 09:49:01.476529 dccex_serial  D2001: TX: <t 1 1 0 0>
2025-03-28 09:49:01.482218 dccex_serial  D2002: RX: <l 1 0 0 0>
2025-03-28 09:49:01.482347 dccex_serial  D2002: RX: <T 1 0 0>
2025-03-28 09:49:01.518029 dccex_network D2002: RX: <l 1 0 0 0>
```
This works very well.

But when it comes to controlling DCC accessory there is no way to know if someone controls an accessory, see log below:
```
2025-03-28 09:49:46.192580 dccex_serial  D2001: TX: <a 1 0>
2025-03-28 09:49:49.371773 dccex_serial  D2001: TX: <a 1 1>
2025-03-28 09:49:52.851653 dccex_network D2001: TX: <a 1 0>
2025-03-28 09:49:53.677656 dccex_network D2001: TX: <a 1 1>
```
It would be really helpful if the other clients are notified, similar to the `<l ....>` response for engine control.

This makes it possible for clients to control accessory, another client (e.g. Traintastic) can then visualize the state of those accessories, e.g. turnout positions.

Asbelos, 2025-03-28:
We do broadcast turnout changes.
That includes turnouts that are not driven by dcc accessory commands. 

If you wish to broadcast specific accessory commands you can intercept them in exrail and broadcast a message of your own design.

reinder, 2025-03-28:
@Asbelos thanks for your response.

> We do broadcast turnout changes.
> That includes turnouts that are not driven by dcc accessory commands.

That requires registration of all accessory turnouts using `<T id DCC addr subaddr>` commands first and then use `<T ..>` to control them. It doesn't really fix the missing broadcast for `<a ...>` and `<A ...>` commands.

I did a quick look into the CommandStation-EX sources, it wasn't that hard to add it :)

[Traintastic](https://github.com/traintastic/traintastic) command log:
```
2025-03-28 22:51:24.830298 dccex_serial D2001: TX: <a 1 1>
2025-03-28 22:51:24.836655 dccex_serial D2002: RX: <y a 1 0 1 1>
2025-03-28 22:51:24.840715 dccex_serial D2002: RX: <y a 1 0 1 0>
2025-03-28 22:51:24.926294 dccex_network D2002: RX: <y a 1 0 1 1>
2025-03-28 22:51:24.926403 dccex_network D2002: RX: <y a 1 0 1 0>
2025-03-28 22:51:29.997321 dccex_network D2001: TX: <a 1 0>
2025-03-28 22:51:30.001467 dccex_serial D2002: RX: <y a 1 0 0 1>
2025-03-28 22:51:30.005615 dccex_serial D2002: RX: <y a 1 0 0 0>
2025-03-28 22:51:30.006324 dccex_network D2002: RX: <y a 1 0 0 1>
2025-03-28 22:51:30.007784 dccex_network D2002: RX: <y a 1 0 0 0>
```